RobTillaart/SHT31 #3: prototype for 'bool SHT31::begin(uint8_t, uint8_t, uint8_t)' does not match any in class 'SHT31'

StefanL38, 2020-07-05:
Hi,

i tried to compile the SHT31_I2Cspeed  example.
The compiler exits with this error-message
`Arduino: 1.8.12 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Stefan\AppData\Local\Arduino15\packages -hardware C:\Users\Stefan\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Stefan\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Stefan\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\Stefan\AppData\Local\Temp\arduino_build_108352 -warnings=none -build-cache C:\Users\Stefan\AppData\Local\Temp\arduino_cache_378192 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.mklittlefs-2.5.0-4-69bd9e6.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.mkspiffs.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -verbose C:\Users\Stefan\Documents\Arduino\SHT31_I2Cspeed-001\SHT31_I2Cspeed-001.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Stefan\AppData\Local\Arduino15\packages -hardware C:\Users\Stefan\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Stefan\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Stefan\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\Stefan\AppData\Local\Temp\arduino_build_108352 -warnings=none -build-cache C:\Users\Stefan\AppData\Local\Temp\arduino_cache_378192 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.mklittlefs-2.5.0-4-69bd9e6.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.mkspiffs.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -verbose C:\Users\Stefan\Documents\Arduino\SHT31_I2Cspeed-001\SHT31_I2Cspeed-001.ino
Using board 'generic' from platform in folder: C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3
Using core 'esp8266' from platform in folder: C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3
Detecting libraries used...
"C:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/include" "-IC:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/lwip2/include" "-IC:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Stefan\\AppData\\Local\\Temp\\arduino_build_108352/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10812 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -DESP8266 "-IC:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3\\cores\\esp8266" "-IC:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3\\variants\\generic" "C:\\Users\\Stefan\\AppData\\Local\\Temp\\arduino_build_108352\\sketch\\SHT31_I2Cspeed-001.ino.cpp" -o nul
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
"C:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/include" "-IC:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/lwip2/include" "-IC:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Stefan\\AppData\\Local\\Temp\\arduino_build_108352/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10812 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -DESP8266 "-IC:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3\\cores\\esp8266" "-IC:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3\\variants\\generic" "-IC:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3\\libraries\\Wire" "C:\\Users\\Stefan\\AppData\\Local\\Temp\\arduino_build_108352\\sketch\\SHT31_I2Cspeed-001.ino.cpp" -o nul
Alternatives for SHT31.h: [SHT31@0.2.1]
ResolveLibrary(SHT31.h)
  -> candidates: [SHT31@0.2.1]
"C:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/include" "-IC:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/lwip2/include" "-IC:\\Users\\Stefan\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.6.3/tools/sdk/libc/xtensa [... cut ...]

   _heatTimeOut = min(180, seconds);

                                  ^

Bibliothek Wire in Version 1.0 im Ordner: C:\Users\Stefan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\Wire  wird verwendet
Bibliothek SHT31 in Version 0.2.1 im Ordner: C:\Users\Stefan\Documents\Arduino\libraries\SHT31  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.`

that's what I love best about "examples"  if not even the given examples compile.

RobTillaart, 2020-07-05:
Good morning Stefan,

I tested the examples mainly with an UNO, but I understand you use them with an ESP8266 module. 
Can you confirm?
Did you test the other examples?
Did that result in the same error message (no need to dump them here right now)

I'll have a look asap

RobTillaart, 2020-07-05:
reverting SHT library to 0.2.1
- selecting sketch - SHT31_I2Cspeed.ino (0.02)
- compiling with UNO works
- compiling ESP32 fails so I can recreate the error

RobTillaart, 2020-07-05:
OK, compiling now works

1) replace void with bool in line 32 of the SHT31.h file
  bool begin(const uint8_t address, uint8_t dataPin, uint8_t clockPin);

2) replace the setHeatTimeout function in SHT31.cpp so it does not use **min()**
```cpp
void SHT31::setHeatTimeout(uint8_t seconds)
{
  _heatTimeOut = seconds;
  if (_heatTimeOut > 180) _heatTimeOut = 180;
}
```

Please give it a try, I will prepare a pull request right away.

StefanL38, 2020-07-05:
Ho Rob,

thank you very much. Now it compiles.

I have another question: What is the right place to ask this question:

Before testing this SHT-library I was using the adafruit-sht-library.
The adafruit-lib needs 500 milliseconds to read temperature and another 500 msecs to read huminity.
Somewhere in the datasheet I read something about that accurate readings need 500 msecs.
Your deomcode I2C-speed reads much more quicker. If I watch the values received there is not much jitter in it.
So does there exist a possability to switch between fastest possible reading and the more accurate reading?

This is not an issue nor a pullrequest so where should I ask questions like this?
best regards Stefan

RobTillaart, 2020-07-05:
Not investigated the latest Adafruit lib but from my head:

When my library doeas a read(), it reads both the T and H does the math for both and puts them into 2 variables. This takes time (~15 millis). When you call getTemperature() or getHumidity() it just copies the variables from the library to your code. This is fast.
To update the internal variables you must call read again, otherwise you get the cached variables again.

The Adafruit library works slightly differently. 
A call to getTemperature() or getHumidity() always reads the sensor again, but when reading the T it ignores the H and vice versa.
So it throws away information e.g. H and most of the time it rereads it immediately again. This results in a double delay (don't know if that is 500 ms but it will do double work. The advantage is that you will have the latest measurement, always.

**Conclusion**
My library stores all fetched information and can deliver info faster as 99% of the cases applications ask temperature and humidity right after each other. If you want my lib to behave like the Adafruit's, you must call read() before every getTemperature() or getHumidity().

RobTillaart, 2020-07-05:
> Ho Rob,
> 
> thank you very much. Now it compiles.
> 
> Your I2C-speed test also explains how to do fast or slow readings
> 
> best regards Stefan

I will merge the pull request, thanks for pointing out the bug and testing.

commit 0d1d8001 (2020-07-05): fix #3 compiling ESP; mismatch begin() + removed min() + add URL in examples.
commit 78318743 (2020-07-05): Merge pull request #4 from RobTillaart/develop

fix #3 Compiling ESP
closed 2020-07-05 by RobTillaart