RobTillaart/SHT31 #20: SHT31 library two I2C interfaces: example, advice needed

tipo1000A, 2021-08-20:
Hi Rob,

I have a custom ATSAMD21G18A based board where I try to read 4 SHT31 sensors connected to 2 I2C interfaces.
I have had no luck with Adafruit_SHT31.h library reading both them (switching back and forth to read).

If I declare _TwoWire myWire(&sercom5, 16, 17);_ , I can read only sensors in the second I2C interface.
If I leave out the declaration, I can read only sensors in default I2C interface.

Your library mentions:
- begin(address, TwoWire *wire) for platforms with multiple I2C busses.

Does it mean I could accomplish the task of reading from 2 I2C interfaces with your library?
Could you give an example how to do it exactly? How to switch back on forth reading with 2 interfaces?

Thanks,
Tipo

RobTillaart, 2021-08-20:
Hi Tipo,

I will make a branch with a test example so we have a "playground to test" OK?

RobTillaart, 2021-08-20:
check out - https://github.com/RobTillaart/SHT31/tree/issue_20

it has an extra example **SHT31_two_I2C.ino** which is straightforward 4 sensors.
I wrote it for an ESP32 that has 2 Wire buses and it compiles, I have no sensors around to test it.

You need to adapt in the top the **TwoWire myWire(&sercom5, 16, 17); ,** declaration 
And rename **Wire1**   to  **myWire**   in two of the four **Begin()** calls. 

Give it a try and please post the result of the compilation and/or if it works.

RobTillaart, 2021-08-20:
updated the example a bit so less editing is needed.

Added a second example using array's (to be tested if the first works.

RobTillaart, 2021-08-20:
just a thought
A possible workaround is using - https://github.com/RobTillaart/TCA9548

RobTillaart, 2021-08-20:
Found - https://learn.adafruit.com/using-atsamd21-sercom-to-add-more-spi-i2c-serial-ports/overview

RobTillaart, 2021-08-24:
@tipo1000A 
Any progress made?

tipo1000A, 2021-08-24:
@RobTillaart 
Hi Rob,

I did try the solution/example you created (thanks), but it didn't seem to work. Then I had to dive to fix another issue on the same board. Now I have that done and I get back to "second I2C" issue.

I did double check with I2C scanner that second I2C I/F (sercom) is indeed working. I also checked with a scope that there is traffic ongoing on those I2C pins.

I'll post you better, more comprehensive info as I'm getting back to that now....

RobTillaart, 2021-08-24:
I have updated the branch, as I released 0..3.3 with a more robust heater code.
Should not change or disrupt the I2C investigation.

tipo1000A, 2021-08-25:
1)	Checking second I2C (SERCOM) functionality with I2C scanner (attached)

To make sure SERCOM I2C is working I connected SHT31 with address 0x44 to default I2C and SHT31 with address 0x45 to SERCOM I2C. I2C scanner output below seems to confirm both I2C’s are working.

```
I2C Scanner

Scanning myWire (sercom5 I2C)...
I2C device found at address 0x45  !
done

Scanning Wire (default I2C)...
I2C device found at address 0x44  !
done

```
[I2C_scanner_SAMD21.ino.txt](https://github.com/RobTillaart/SHT31/files/7044988/I2C_scanner_SAMD21.ino.txt)

tipo1000A, 2021-08-25:
2)	Having confirmed the above I added other SHT31 sensors' so that both I2C's have 0x44 and 0x45, and then I tried to run your sketch with couple of edits to match my board.

The only output was this:

`SHT31_LIB_VERSION: 	0.3.3 `

Checking with scope at the same time I didn't see any traffic in SERCOM I2C. In default I2C I saw small some traffic one time, then no traffic.
[KUI_NO_LoRa_ROB_I2C_orig.ino.txt](https://github.com/RobTillaart/SHT31/files/7045050/KUI_NO_LoRa_ROB_I2C_orig.ino.txt)

tipo1000A, 2021-08-25:
3. If I comment out this:

```
 b1 = sht_1.begin(SHT31_ADDRESS_1, &Wire);
  b2 = sht_2.begin(SHT31_ADDRESS_2, &Wire);
 // b3 = sht_3.begin(SHT31_ADDRESS_3, &myWire);
 // b4 = sht_4.begin(SHT31_ADDRESS_4, &myWire);
```
I get this output:

```
SHT31_LIB_VERSION: 	0.3.3
BEGIN:	1	1	0	0	
23.8	22.8	-45.0	-45.0	61.1	66.7	0.0	0.0
23.9	22.8	-45.0	-45.0	61.1	66.5	0.0	0.0
23.8	22.8	-45.0	-45.0	61.1	66.2	0.0	0.0
23.8	22.9	-45.0	-45.0	60.9	66.2	0.0	0.0
23.9	22.8	-45.0	-45.0	61.1	66.1	0.0	0.0
```

RobTillaart, 2021-08-25:
And if you commented out b1 and b2  the other pair worked?

tipo1000A, 2021-08-25:
With this:

```
  //b1 = sht_1.begin(SHT31_ADDRESS_1, &Wire);
  //b2 = sht_2.begin(SHT31_ADDRESS_2, &Wire);
  b3 = sht_3.begin(SHT31_ADDRESS_3, &myWire);
  b4 = sht_4.begin(SHT31_ADDRESS_4, &myWire);
```

The output is just this:

`SHT31_LIB_VERSION: 	0.3.3`

So, didn't work...

RobTillaart, 2021-08-25:
rereading - https://learn.adafruit.com/using-atsamd21-sercom-to-add-more-spi-i2c-serial-ports/creating-a-new-wire

and looked at this datasheet - https://www.microchip.com/en-us/product/ATSAMD21G18

you use pin 16 and 17.  Can you explain why you choose those two pins?

RobTillaart, 2021-08-25:
If I understand the theory correctly I would try sercom4 and pin 19, 20

[image]
(page 29 datasheet)

You might notice that pin 17 and 18 are skipped in that column.

RobTillaart, 2021-08-25:
OK found where the numbers 16 and 17 possibly come from.
[image]
(page 33 datasheet)

Looking up these in the MUX table. 

[image]
(page 30 datasheet)

It looks like you need to use sercom3, and pins 25 and 26 ?

```cpp
TwoWire myWire(&sercom3, 25, 26);

#include "wiring_private.h" // pinPeripheral() function
...
  pinPeripheral(25, PIO_SERCOM);
  pinPeripheral(26, PIO_SERCOM);
```

tipo1000A, 2021-08-25:
The history is that before I started to make the layout for my custom board I prototyped the use of physical pins #47 and #48 on the chip. I did that with I2C scanner and it was working (still is) and so I chose to use those pins. They were also most convenient to use from PCB layout perspective.

Physical pins #47 and #48 translate to I/O pins PB02 and PB03 respectively. From datasheet table 7-1.

[image]

[image]

The pin numbers to use with "pinPeripheral" (16 and 17) I picked up from variants.cpp file of MKRWAN1300. [https://github.com/arduino/ArduinoCore-samd/blob/master/variants/mkrwan1300/variant.cpp](url)

[image]

I thought I did everything right since I was able to ping sensors with  myWire.beginTransmission(address);
    error = myWire.endTransmission(); with I2C scanner. It must mean that the right "port" can be opened and physically it works, right?

Can you explain why I'm able ping sensors with  myWire.beginTransmission(address);
    error = myWire.endTransmission(); but can't actually communicate with them using library functions?

RobTillaart, 2021-08-29:
Any progress made?

tipo1000A, 2021-09-01:
I think I'll have update on this on Monday 6.9.

RobTillaart, 2021-09-15:
Any progress?

tipo1000A, 2021-09-18:
Hi,

Using MKR WAN1310 board I tested SHT31 on pins PA22 (SDA) and PA23 (SCL) and it works.

`TwoWire myWire(&sercom5, 0, 1);`

```
  pinPeripheral(0, PIO_SERCOM_ALT);
  pinPeripheral(1, PIO_SERCOM_ALT);
```

I tested with one SHT31 (0x45) on this SERCOM5 interface. I still need to connect a sensor to default I2C I/F and make sure they both work in the same sketch.

tipo1000A, 2021-09-18:
I hooked up another SHT31 to default I2C and they both (default and SERCOM) work fine!

Thanks for putting your time and effort on this. In the end the whole thing was because I didn't read the spec carefully.

RobTillaart, 2021-09-18:
> In the end the whole thing was because I didn't read the spec carefully.

Reading specs is hard you are definitely not alone in that one.

Can you post your final sketch that works?
I would like to add it to the examples in the library.

closed 2021-09-20 by RobTillaart