RobTillaart/SHT31 #23: Return 0.0C and 0.0%

Lucamenghini, 2021-09-12:
Hi, I'm using an Heltec Wireless Stick (ESP32 LoRa, wifi, ble etc).
I wired the sensor and launched the SHT31_IsCOnnected.ino but the result is 
23:08:36.642 -> 194035	160	0.0	0.0
23:08:37.644 -> 195035	160	0.0	0.0
23:08:38.644 -> 196035	160	0.0	0.0
23:08:39.644 -> 197035	160	0.0	0.0
23:08:40.644 -> 198035	160	0.0	0.0
23:08:41.648 -> 199035	160	0.0	0.0
23:08:42.660 -> 200035	160	0.0	0.0
23:08:43.660 -> 201035	160	0.0	0.0
23:08:44.661 -> 202035	160	0.0	0.0
23:08:45.661 -> 203035	160	0.0	0.0

what I wrong?
Thank you, bye

Lucamenghini, 2021-09-12:
Now I've updated to V.0.3.3 and the result of the SHT31_IsConnected is
00:04:14.254 -> SHT31_LIB_VERSION: 	0.3.3
00:04:14.254 -> 8010
00:04:14.254 -> 40	4115	-45.0	0.0
00:04:15.243 -> 1044	3989	-45.0	0.0
00:04:16.264 -> 2048	3989	-45.0	0.0
00:04:17.255 -> 3052	3989	-45.0	0.0
00:04:18.266 -> 4056	3989	-45.0	0.0
00:04:19.291 -> 5060	3989	-45.0	0.0
00:04:20.285 -> 6064	3989	-45.0	0.0

RobTillaart, 2021-09-13:
Hi,
There are a number of possible problems, I always start my analysis bottom up, so at the hardware 

Do you have pull up resistors? 
What is the cable length?
Does the sensor get enough voltage?

----

**SHT31_IsCOnnected.ino**

The status returned = **8010**  ==>  https://github.com/RobTillaart/SHT31/readme.md  (or datasheet)

This is the default value of the status register so it looks like you make the connection correctly 
(change 1 in 65536 that you got this value just by random noise)

```
00:04:14.254 -> 40 4115 -45.0 0.0
00:04:15.243 -> 1044 3989 -45.0 0.0
00:04:16.264 -> 2048 3989 -45.0 0.0
00:04:17.255 -> 3052 3989 -45.0 0.0
```
The time between measurements is about 1 second and the read time is stable. 
However it looks like there is some error during the read(), lets try to catch it.

Can you change **loop()** to

```cpp
void loop()
{
  if ( sht.isConnected()  )
  {
    start = micros();
    bool b = sht.read();         // default = true/fast       slow = false
    stop = micros();

    int error = sht.getError();
 
    Serial.print(b, HEX);
    Serial.print("\t");
    Serial.print(error, HEX);
    Serial.print("\t");
    Serial.print(millis());
    Serial.print("\t");
    Serial.print(stop - start);
    Serial.print("\t");
    Serial.print(sht.getTemperature(), 1);
    Serial.print("\t");
    Serial.print(sht.getHumidity(), 1);
  }
  else
  {
    connectionFails++;
    Serial.print(millis());
    Serial.print("\tNot connected:\t");
    Serial.print(connectionFails);
    // sht.reset();
  }
  Serial.println();
  delay(1000);
}
```

RobTillaart, 2021-09-20:
@Lucamenghini 
Any progress to report?

RobTillaart, 2021-09-30:
As there is no reaction I assume problem is solved, otherwise reopen the issue and provide more information.

closed 2021-09-30 by RobTillaart

Lucamenghini, 2022-01-23:
Hi i solved using the sensor on other declared pins for I2C

#define PIN_SDA2 18 //  SDA SECOND I2C SHT30
#define PIN_SCL2 19 // SCL SECOND I2C SHT30

I didn't understand why they could not use the same pins for I2C with different address and same clock....
But forcing sensor on other PINs declaring a second I2C then problem is solved.

The air baud rate is setted very slow otherwise 50% packets were lost
thank you
bye