RobTillaart/SHT31 #26: SoftWire library + SHT31 library

SanSanysch, 2021-11-04:
Hi, Rob!
I have: Arduino ATmega2560, 4 pcs SHT31.
2 sensors (addr. 0x44 and 0x45) connect to hardware I2C, with you library and wire,
2 senors (addr. 0x44 and 0x45) connect to software I2C, with library https://github.com/felias-fogg/SoftI2CMaster (SoftWire.h)
Everything works fine! But ...I can't make friends SoftWire + SHT31 libraries, to use all the charms of your library.
You could contact the author: https://github.com/felias-fogg and develop a joint decision?
Thank you in advance.

RobTillaart, 2021-11-05:
@SanSanysch 
Will look into this latér today. 
Can you post a stripped minimal version of your code?
Can you tell what the error message is?

RobTillaart, 2021-11-05:
From the i2c lib you use

```
If you want a solution running on an ARM MCU (Due, 
Zero, Teensy 3.x), you want to use pins on port H or 
above on an ATmega256, or you want to use many 
different I2C buses, this library is not the right 
Solution for you.
```
So it might be a pin problem.

Have you tried other pins for the sw i2c?

RobTillaart, 2021-11-05:
@SanSanysch 
are you familiar with https://github.com/Testato/SoftwareWire ?

Not tested it myself (yet) but some of the contributers are involved with Arduino core.

SanSanysch, 2021-11-05:
Thanks for fast answer. Minimal code:
`// SHT31 on hardware I2C
#include "Wire.h"
#include "SHT31.h"
SHT31 sht_a;
SHT31 sht_b;

// SHT31 on software I2C
#define SDA_PORT PORTD
#define SDA_PIN 3			// = D18
#define SCL_PORT PORTD
#define SCL_PIN 2			// = D19
#include <SoftWire.h>
SoftWire SWire = SoftWire();

void setup(void) {
  Serial.begin(9600);
  Wire.begin();
  sht_a.begin(0x44);
  sht_b.begin(0x45);
  SWire.begin();
}

void SHT31(byte ADDR)
{
  byte data[6];
  SWire.beginTransmission(ADDR);
  SWire.write(0x2C);  // Send measurement command
  SWire.write(0x06);
  SWire.endTransmission(false);
  SWire.requestFrom(ADDR,6);
  if (SWire.available() == 6) { 
    data[0] = SWire.read();    // cTemp msb
    data[1] = SWire.read();    // cTemp lsb
    data[2] = SWire.read();    // cTemp crc
    data[3] = SWire.read();    // humidity msb
    data[4] = SWire.read();    // humidity lsb
    data[5] = SWire.read();    // humidity crc
    }
  // Convert the data
  float cTemp = ((((data[0] * 256.0) + data[1]) * 175) / 65535.0) - 45;
  float humidity = ((((data[3] * 256.0) + data[4]) * 100) / 65535.0);

  // Output data to serial monitor
  Serial.print(ADDR);
  Serial.print(" Humidity: ");
  Serial.print(humidity);
  Serial.print(" %RH");
  Serial.print("\t");
  Serial.print(" Temperature: ");
  Serial.print(cTemp);
  Serial.println(" C");
}

void loop()
{
  sht_a.read();         // default = true/fast       slow = false
  Serial.println("===hardware I2C===");
  Serial.print("sht_a");
  Serial.print(" Humidity: ");
  Serial.print(sht_a.getHumidity(), 1);
  Serial.print(" %RH");
  Serial.print("\t");
  Serial.print(" Temperature: ");
  Serial.print(sht_a.getTemperature(), 1);
  Serial.println(" C");

  sht_b.read();         // default = true/fast       slow = false
  Serial.print("sht_b");
  Serial.print(" Humidity: ");
  Serial.print(sht_b.getHumidity(), 1);
  Serial.print(" %RH");
  Serial.print("\t");
  Serial.print(" Temperature: ");
  Serial.print(sht_b.getTemperature(), 1);
  Serial.println(" C");

  Serial.println("===software I2C===");
  SHT31(0x44);
  SHT31(0x45);
  Serial.println();
  delay(2000);
}`
Yes i familiar with https://github.com/Testato/SoftwareWire but not check.
Unfortunately, I will not at home until Sunday. Please, wait.

RobTillaart, 2021-11-05:
updated your post to do syntax highlighting.

RobTillaart, 2021-11-05:
I think the SoftWIre library blew up my SHT85 sensor which I used for testing.  (stupid me :( )

RobTillaart, 2021-11-05:
> @SanSanysch are you familiar with https://github.com/Testato/SoftwareWire ?
> 
> Not tested it myself (yet) but some of the contributers are involved with Arduino core.

you should have the v1.5.1 as this one inherits from TwoWire.

SanSanysch, 2021-11-07:
I checked https://github.com/Testato/SoftwareWire - it work. I replace library in my example above, but i did not understand, how to bind you library and Softwire-library. In your library hard used Wire-library. I don't want edit your library.

SanSanysch, 2021-11-07:
I found possible solution: https://github.com/Testato/SoftwareWire/wiki/Using-a-library-that-needs-a-TwoWire-pointer
may be your edit library for versatility. I'm sure, it's useful for someone.

RobTillaart, 2021-11-07:
There are two strategies discussed:

**If the Arduino Wire library is not used at all**

As more and more sensors / IC / display  are available with I2C this scenario is unlikely. At least from a library point of view I just don't know if the user of my lib does or does not use more I2C devices. Looking at my own projects almost all use I2C character displays, sensors and PCF8574 (IO) for actuators. In fact on PCB's I always want one or two I2C connectors to extend functionality in the future (without redesigning main board).

So for me this is no option.

**If the Arduino Wire library is also used**

This is the safest assumption from library point of view.  If I understand the example it is a "copy" of the LiquidCystal display library which has all TwoWire removed and SoftWire put in place instead. That would imply that maintenance for the library increases. I estimate minimal 30% extra time (especially testing).  As I have now ~130 libraries from which 80+ uses Wire that would substantially increase the work. Yes I know it is useful for somebody (that's is why I share my libs) but please understand that it costs me substantial free time that I cannot spend on my own projects.

> I will check, if you do not have sensors.

Thank you, I have ordered 2 SHT85's and I expect them end of this week (not confirmed)

SanSanysch, 2021-11-07:
> Thank you, I have ordered 2 SHT85's and I expect them end of this week (not confirmed)
OK. I'll wait. Let me know about the results.

RobTillaart, 2021-11-08:
Got order confirmation saying that the sensors are expected in 3rd week januari 2022.
So to be continued when (if) they arrive over 10 weeks or so.
Sorry for the inconvenience.

SanSanysch, 2021-11-08:
There did you bay sensors? To us, from China, comes in about a month.
There are no other options. Will wait...

RobTillaart, 2021-11-08:
Ordered them from rs-online, normally they deliver in 2 or 3 days.

SanSanysch, 2022-01-26:
Hi, Rob! You received sensors?

RobTillaart, 2022-01-26:
No, nothing, not even a message that they will be late ...

RobTillaart, 2022-03-23:
Had a lot of time to rethink the question and I decided to not change the library for this question. 
Any **softWire** library that implements the **TwoWire** interface can be used with the SHT31 lib,
so there is a provision by means of dependency injection. 

It is not my goal to maintain separate versions of libraries that are tuned to specific use-cases / projects.
The MIT license allows you to fork the library and patch it to whatever is needed and even use it commercially.

[image]

At the moment I have no concrete plans to write a **TwoWire** compatible softwire library, so that route will not be a part of a solution for your question. So I will close this issue.

FYI, the sensors are still not available :(

SanSanysch, 2023-04-04:
It's very good news! Thank you very much!!!

RobTillaart, 2023-04-04:
Give it a try!

SanSanysch, 2023-05-22:
Hi, Rob! I check sensors SHT31 on Arduino atmega 2560 with SHT31_SW + SoftwareWire - it works! And very fast! I replace SoftWire library on my PC. Unfortunately with SoftWire - don't work. Thank you for good job!
I think, you need correct SHT31_SW library finally with SoftwareWire library.

RobTillaart, 2023-05-23:
@SanSanysch 
Thanks for testing, that confirms my tests.
Hope to find some time this week.

RobTillaart, 2023-07-20:
@SanSanysch 
Created a separate repo with an initial version with SoftwareWire 

- https://github.com/RobTillaart/SHT31_SWW  - 

There are problems with build, but local build with 1.8.19 IDE and UNO works.