RobTillaart/SHT31 #31: Using SHT31 in High Humidity Environment

chinswain, 2022-05-03:
I'm planning on using an SHT31 in an environment that's 70 - 100% humidity, I was going to leave the heater on permanently to keep condensation away from the sensor (To both extend its lifespan and provide more accurate humidity readings) and then calculate/compensate for the increased temperature and decreased humidity in code, but I see you have a warning about leaving the heater on for long periods.

Do you have any recommendations (in your experience) with using the heater long term? Is there a minimum heat up and cool down period before taking readings?

RobTillaart, 2022-05-04:
Interesting question,  Unfortunately I have no answer as i did not test high humidity environments.
The limits are based on the datasheet, expect they exist to prevent damage.

(Some thoughts to consider, use at own risk)
First I would run two sensors side by side, one heating and one without, just to study the effect of the constant heating. Preferably in a medium humidity range first.

Second I think i would not use constant heating but a sort of pulse modulation dependant on the humidity.. e.f. range 0-50% duty cycle mapped from 70-100% humidity. Heating cycles of 0..5 seconds followed by 5 seconds cooling. Maybe even shorter to keep the added energy / temp sort of constant

RobTillaart, 2022-05-08:
@chinswain
As this topic is better discussed on e.g. the Arduino forum I close the issue. (it is not sec a library issue)

If you gain some insights and you want to have them added to the documentation of the library, or if you created a specific example for this, please reopen the issue.

closed 2022-05-08 by RobTillaart

RobTillaart, 2022-05-09:
That looks very good, 
- humidity stable around 92%  (I assume)
- temperature about 25°C (I assume)

works beyond my expectations, I expected to see the pulses of the heater back in the humidity or temperature.
👍

RobTillaart, 2022-05-10:
Great info!
Could you use the value of H as trigger to start / stop the heater?

chinswain, 2022-05-10:
I thought about that, but there's no information on what level would require the heater, I guess I could calculate the dew point and heat to prevent condensation. I'm assuming the membrane gets saturated and somehow damaged over time. I rarely get 6 months out of sensors (BME280 etc) so hopefully with periodic heating it'll last a bit longer! I've setup 5 of them on an ESP32, one on each terrarium and another outside.