RobTillaart/SHT31 #33: Heater

jomardyan, 2022-08-16:
Hi, how to effectively use a heater in a real environment? Should it be full time on?

RobTillaart, 2022-08-17:
Thanks for the question.
From my head, no it should not as the heater would affect the measurements 
I need to check the datasheet to verify as it has been a while i read that one.

RobTillaart, 2022-08-17:
Read some sources:

The SHT31 datasheet is quite silent about it, so had a look in related datasheets / sources

#### SHT85 datasheet 
_The SHT85 is equipped with an internal heater, which is meant for plausibility checking only. The temperature increase
achieved by the heater depends on various parameters and lies in the range of a few degrees centigrade._

#### SHT21 datasheet 
_The heater is intended to be used for functionality diagnosis – relative humidity drops upon rising temperature._

#### Adafruit
https://learn.adafruit.com/adafruit-sht31-d-temperature-and-humidity-sensor-breakout?view=all
_There's also a heater built into the sensor, used to heat/evaporate any condensation._

So the purpose is to check if the sensor is working. => Temperature should rise and humidity should drop.
Therefor the heater should normally be OFF (default), I do not recall ever using the heater except for testing the library.

The Adafruit site mentions condensation removal, however I did not see that in the datasheet or other source.

The heater should be on for max 180 seconds, however I cannot find the source of that statement.

The library does support heater control, but the user is responsible for timing it.

RobTillaart, 2022-08-22:
@jomardyan 
If this solves the question you may close the issue

jomardyan, 2022-08-22:
It's better to enable discussion section :)

closed 2022-08-22 by jomardyan