RobTillaart/SHT31 #41: sht.begin() no answer

tk5ep, 2025-03-26:
Hi,

When the SHT31 is connected to SDA and SCL but not powered, the driver is locked when calling sht.begin().
No answer when calling sht.readStatus().

Removing the SHT31 unlocks the situation and the status is FFFF.

Is there a way to correct this ?

Thanks,

RobTillaart, 2025-03-26:
Thanks for your question,

mmm, there is an timeout option for I2C if I recall correctly, don't know the details from my head.

RobTillaart, 2025-03-26:
@tk5ep 

https://docs.arduino.cc/language-reference/en/functions/communication/wire/setWireTimeout/

Give this a try

tk5ep, 2025-03-26:
Hi Rob,
That worked !

Thanks for the help and all your nice libraries. Great job ! 👍

RobTillaart, 2025-03-26:
@tk5ep 

Just curious, what value did you use as timeout?

As the question is answered you may close the issue.

closed 2025-03-26 by tk5ep